GEECS-BELLA/GEECS-Plugins #159: change where uint16 is enforced in ImageAnalysis workflow

skbarber, 2025-05-13:
the build_return_dictionary is a method of ImageAnalysis that is used to structure returns from ImageAnalysis.analyze_image() in a semi-standard way. It's not required to be used by ImageAnalysis, but it is a useful shortcut. It was originally designed for use with ImageAnalyzers used for Labview, which is very strict with the return type, but it was added to the base class of ImageAnalysis. ImageAnalysis itself is more flexible with return types. In any case, I would like to change this part of the method which currently coerces the returned image to the uint16 type. 
https://github.com/GEECS-BELLA/GEECS-Plugins/blob/d310f360b7561e3105a289a6cd666e03e4bbc510/ImageAnalysis/image_analysis/base.py#L178

While this is necessary for the labview workflow, it can be enforced further downstream, for example, in labview_adapters.py:
(https://github.com/GEECS-BELLA/GEECS-Plugins/blob/d310f360b7561e3105a289a6cd666e03e4bbc510/ImageAnalysis/image_analysis/labview_adapters.py#L113C1-L114C1)

Reorganizing in this way allows ImageAnalysis to return non uint16 results, which is useful for things like returning background subtracted images that have potentially negative values. In principle, the negative values could be dealt with by an offset, but that would require tracking of additional information. Overall, allowing a bit more flexibility in the return type is useful and I think the uint16 requirement for labview can still be preserved. In fact, the conversion to uint16 should definitely be done in the ImageAnalyzer before sending as a return anyways. Probably, it makes sense to log a warning or error in the case that a non-uint16 image is being returned for use in labview rather than coercing it to 16 bit, which could lead to hard to track down errors.

cedoss-lbl, 2025-05-14:
I can work on a fix for this, but should I do this change on a fresh branch off of master or use the new pull request #160 and make this change on that branch?

skbarber, 2025-05-14:
I would say it makes sense to make a fresh branch off master and work on
this there. I don't think there will be any compatibility issues

skbarber, 2025-05-14:
I would think there isn't a huge amount of code that needs to be modified
(though, we may want to rename 'processed_image_uint16' key, which would
just need to be traced carefully)

cedoss-lbl, 2025-05-14:
After looking at these two files more I agree with you.  Seems like it makes the most sense to shift the uint16 enforcement to `labview_adapters.py`.  I'll make a PR shortly with the changes.  (Which will be mostly replacing the key name in Array2DScanAnalysis)

The one comment I'll make is that I don't think we need to make a warning if the user is pushing non-uint16 images up to labview_adapters.  We can just have a `image=image.astype(np.uint16)` to take care of the conversion in the backend so people don't need to explicitly do this in their analyzers.

skbarber, 2025-05-15:
looks good, closing issue

closed 2025-05-15 by skbarber